openego/ding0 #5: handling of projections

nesnoj, 2016-05-11:
Why did you add pyproj to reproject geom data from DB? (ca90b58)
The reprojection to WGS84 is already done in the SQL statement.

For future coding (after switching to SQLAlchemy) we have to rediscuss which module we ll use for reprojections, pyproj or [GeoAlchemy](http://geoalchemy-2.readthedocs.io/en/0.2.6/index.html) (SQLAlchemy doesn't support PostGIS operations except in raw sql statements)

gplssm, 2016-05-12:
Hard to remember, but i got it ;)
I had to transform to srid 3035 to obtain calculated area of mv_region in m^2 instead of square degrees.

gplssm, 2016-05-30:
Is the question answered? Can we close the issue?

gplssm, 2016-06-29:
Decision: If small re-projections are required in Dingo, pyproj as applied in ca90b585397ced3e2524178bbcc0e07e17dae0d8 is the right choice.

This issue/the greater topic is related to #10 and the creation of [ego.io](https://github.com/openego/ego.io). When you open #10, find a solution for re-projection because this can't be done in ego.io due to maybe different required output projections.

closed 2016-06-29 by gplssm