openego/ding0 #8: passing unnecessary args to a function/method

nesnoj, 2016-05-17:
Example in class [MVGridDingo](https://github.com/openego/dingo/blob/dev/dingo/core/network/grids.py#L95-L96): Here 2 args are passed that can be easily obtained within the method _parametrize_lines_ instead. Which is the appropriate way?

Args are obtained within method instead:
**Pros:** Clearer code
**Cons:** No documentation for the arguments within docstring

MGlauer, 2016-05-17:
This highly depends on whether the method should be called with parameter other than the `self.region.peak_load` and `self.region.geo_data.area`. 

If *yes*: Everything is fine. But: both methods contain `self.region`. Hence, it might be possible that both parameters origin from different regions, i.e. 
`parametrize_lines(regionA.peak_load, regionB.geo_data.area)'
Is this a use case? You might want to pass the region object instead of both attributes.

If *no*: This method does not need any parameters.

nesnoj, 2016-05-18:
Thank you for your reply Martin, good to know you're on the watchlist! :)
Actually, different regions are not a use case so I'd go as well for the no-parameters-way..

commit 8f6c0935 (2016-05-18): remove parameters of method 'parametrize_lines', obtain them within the method instead. Closes #8
closed 2016-05-18 by nesnoj