openego/ding0 #9: parametrization of line/cables

nesnoj, 2016-05-17:
2 comments on MVGridDingo class' method [parametrize_lines](https://github.com/openego/dingo/blob/dev/dingo/core/network/grids.py#L99):

1. You do the parametrization after the routing but line specs are required to [check technical constraints](https://github.com/openego/dingo/blob/dev/dingo/grid/mv_routing/models/models.py#L172) during routing process - let's discuss this topic again..?
2. For iteration of the graph's edges ([used here](https://github.com/openego/dingo/blob/dev/dingo/core/network/grids.py#L167)), you can use the generator method [GridDingo.graph_edges()](https://github.com/openego/dingo/blob/dev/dingo/core/network/__init__.py#L64)

nesnoj, 2016-05-18:
I'll move the method to the tech. constraints

gplssm, 2016-05-18:
Point (1) is solved by moving and adapting parametrize_lines().

nesnoj, 2016-08-01:
Done

closed 2016-08-01 by nesnoj