openego/ding0 #10: use SQLAlchemy for DB access

nesnoj, 2016-05-18:
Is required since @MGlauer uses SQLAlchemy for the interface to oedb.
We need [GeoAlchemy](http://geoalchemy-2.readthedocs.io/en/0.2.6/index.html) for PostGIS types & operations.

nesnoj, 2016-05-25:
I added SQLAlchemy table definitions for import & export in 0451178.

Next step: Change import methods in [NetworkDingo](https://github.com/openego/dingo/blob/dev/dingo/core/__init__.py#L13). Since there are no operations allowed in the table definitions mentioned above, the calculations and joins within the old SQL statement have to be replaced by SQLAlchemy operations..

Question: Since we gonna use the new SQLAlchemy format for DB-interface description, there's no need for the [config_db_tables](https://github.com/openego/dingo/blob/dev/dingo/config/config_db_tables) anymore, right?

gplssm, 2016-07-06:
Finished by 19fb0b78737834b1245cf5e69ce26ba2275c7b46

closed 2016-07-06 by gplssm