openego/ding0 #21: check PyPSA class model for use within DINGO

nesnoj, 2016-05-18:
Get deeper understanding of [PyPSA's](http://www.pypsa.org/doc/index.html) and [oemof.powerflow](https://github.com/oemof/oemof) class models and answer following questions:

1. Is a subset of classes suitable for use within DINGO (e.g. elec. components only) ?
2. What is the target format of DINGO's grid - oemof.powerflow or PyPSA ?
3. ...?

@gplssm, any additions?

gplssm, 2016-05-30:
1. [The wiki](https://github.com/openego/dingo/wiki/Class-refactoring) contains some comparison of current used Dingo classes with PyPSA classes (currently not used classes of dingo, such as generator, load, bus) are not treated there. I would recommend to used generator, load and bus directly from PyPSA and translate current TransformerDingo and BranchDingo to PyPSA classes during the _PyPSA_ translation process_
2. Due to scarce implementation progress and missing clear final structure I would recommend to build dingo directly against PyPSA (for now).

gplssm, 2016-06-13:
PyPSA classes won't be used in Dingo directly. When needed, a translator to a PyPSA problem will be written to enable Dingo to make power flow analysis for checking technical constraints.

gplssm, 2016-06-29:
We use an exporter for the whole grid created by Dingo. #34 replaces this issue.

closed 2016-06-29 by gplssm