openego/ding0 #23: using parent's class attribute within subclasses - how to ensure distinct attribute types?

nesnoj, 2016-05-30:
Supposed I got a class `Grid` with an attribute `Region` and two child classes `MVGrid` and `LVGrid`. The child classes store different types in the inherited attribute, more precisely `MVGrid.Region = some MVRegion object` and  `LVGrid.Region = some LVRegion object`. Thus the attribute has no distinct type (and makes it hard to draw in UML class diagram).

To avoid this ambiguity I'd remove this attribute in parent class and add it to `MVGrid` as well as to `LVGrid`. Is there are better way to handle this?

nesnoj, 2016-05-30:
addition: in UML class diagram I could associate the parent class `Region` as attribute but since the attribute contains an instance of a subclass of `Region` it's not correct, is it?

MGlauer, 2016-05-30:
This sounds like a scenario where you want to use the "redefines"-Tag in UML:
[image]

nesnoj, 2016-05-30:
Thanks for your reply!
But what do I need the region attribute in the parent class for then? It's not used elsewhere.. It might be "cleaner" to just declare it as an attribute in the subclasses?!

MGlauer, 2016-05-30:
This depends on whether a region should be part of a (general) grid. Suppose you want to implement some general functionality that needs this property (e.g. printRegion()). 
Normally, it is way cleaner if we do not omit this attribute. And a shared attribute in subclasses is a strong indicator that it should be present in the super class.

nesnoj, 2016-05-31:
Currently there's no intention to implement more general functionality but you never know...so I'll follow your recommendation - cheers!

closed 2016-05-31 by nesnoj