openego/ding0 #26: type adaption in sqla/psycopg2 to match SQL types

nesnoj, 2016-05-31:
[Adding a dataset to a table](https://github.com/openego/dingo/blob/dev/dingo/core/__init__.py#L305) via sqla throws:
`sqlalchemy.exc.ProgrammingError: (psycopg2.ProgrammingError) can't adapt type 'numpy.int64'`
(variable `grid_id` is of type 'numpy.int64') so I seems psycopg2 doesn't know how to map the numpy to SQL-conform type (column type is defined as 'sqla.Integer' [here](https://github.com/openego/dingo/blob/dev/dingo/config/config_db_interfaces.py#L89)).

I found a [solution](http://initd.org/psycopg/docs/advanced.html#adapting-new-python-types-to-sql-syntax) using a psycopg2 extension using `register_adapter()`, (shabby) implementation see commit 4469af6. Is this the right way to handle the type adaption or is there a more elegant one?

MGlauer, 2016-06-01:
I think this is ok. But the registration should not be part of a method but of the package. Even better: It should be part of the planned ego.io-packages `__init__.py`, i.e. if you use `import ego.io` this adapter will be automatically registered.

nesnoj, 2016-06-01:
I agree, my version is for testing purposes only..
I also thought about moving it either to ego.io or to oemof.db (in case they switch to sqla)..

gplssm, 2016-07-06:
With 87eba5bfcd544adcc0f521a92c2c7d58219b4791 register_adapter is moved to Dingo's root init. I'm not sure if this is a good place. Once, we have agreed on a structure for ego.io we can move there.

gplssm, 2016-09-14:
Check if the `register_adapter` is sufficient to solve this issue!

Would be nice, cause this would solve for all packages that import `ego.io`

gplssm, 2016-11-10:
I think I solved this issues by https://github.com/openego/ego.io/issues/1

@MGlauer could you please close this issue if you accept the solution?

closed 2017-04-11 by nesnoj