openego/ding0 #33: avoid double feed in areas near grid district boundary

nesnoj, 2016-06-16:
Some load areas are cut by grid district borders, both got peak loads more than the min. peak load threshold. Therefore each of them is fed by a different MV grid - this results in overestimation of line length.

Example:
[image]

nesnoj, 2016-06-29:
Reason was a bug in load area SQL script.
@gplssm, please recalculate peak loads.

gplssm, 2016-07-06:
I would like to, but I can't because materialized views depend on the table and I can't delete it. What's the plan?

nesnoj, 2016-07-07:
Make @ludee set the correct permissions :)

gplssm, 2016-07-07:
Done!

nesnoj, 2016-07-15:
Addition:
Peak loads are correct right now - some of the regions now got peak loads smaller than threshold, hence less regions are double fed, but problem still existent. It is caused by the grid district method ("cut and forget").
We have to discuss how we deal with this bug - input: In reality, there are interconnections between different MV grids. The present bug doesn't connect the MV grids but leads to longer lines that could kind of represent the real "length-overhead" of the lines. But we actually can't prove it..

gplssm, 2016-12-23:
After _data processing_ was revised, check if this issue is still present

closed 2017-09-05 by nesnoj