openego/ding0 #40: pick up the salt shaker

nesnoj, 2016-07-08:
Parts of Dingo are not deterministic - for different runs with same input data, different grids are generated. So let's get back control of the salt shaker to gain explicit instead of implicit salt!

nesnoj, 2016-07-21:
I found it!
..cleaned up the salt in 68c54c5.
closed.

gplssm, 2016-07-21:
This solution is valid for a static input data set. But, if something changes there resulting MV grids generated by Dingo may be different.

On the example of changed IDs in the load_area data set:
- It is probable that load_area IDs will change if for example new MV stations are added
- This may lead to different results obtained from the routing in Dingo

Either clearly state that MV grids generated by Dingo belong to one particular input data (or are a part of this data set) or try to keep all IDs constant in the datasets prior to Dingo.
Alternative or: try to iterate over a different column than id within Dingo.

nesnoj, 2016-12-21:
**Current status:**

1. Topology of MV grid including strings diverting from main ring is deterministic.
2. The major part of the MV topology within LV load areas seems to be deterministic (checked with >10 runs). However, in some larger LV load areas there are small parts that differ for several runs.
Example:

_Run 1:_
[image]
_Run 2:_
[image]

This behaviour is related to the methods in [mv_connect_satellites()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/grid/mv_grid/mv_connect.py#L483) and [mv_connect_stations()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/grid/mv_grid/mv_connect.py#L560) (called in [MVGridDingo.routing()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/core/network/grids.py#L207) for every MV grid). If one of the two functions mentioned above is omitted during execution the grid seems to be fully deterministic hence the combination of both functions seem to spill some salt..
Detail from diving into mv_connect_stations(): During connection of stations within a load area, the nearest connection objects for every station are determined using [find_nearest_conn_objects()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/grid/mv_grid/mv_connect.py#L660). The returned stack changes for different runs.

**Conducted measures:**
- The topology of a grid strongly depends on the order in which all single steps are conducted. One important measure is to avoid implicit "salt" (seemingly random results), e.g. by sorting the results of a generator prior to further use. Therefore, in (hopefully) all necessary parts of the code this basic idea was considered. Example: e0c410d60e3bd99e26027133693835260e40800c
- To compare the result (MV grid's graph) of two different runs there's a basic function [compare_graphs()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/tools/debug.py#L8). It can save a graph (using `mode='write'`) as [gpickle](http://networkx.readthedocs.io/en/stable/reference/readwrite.gpickle.html) and compare (using `mode='compare'`) the graph of a second run with the saved one. The comparison is done in a very basic manner and does only include nodes and edges. NetworkX provides some [functions](http://networkx.readthedocs.io/en/stable/reference/algorithms.isomorphism.html) to check if two graphs are isomorphic but it fails to hande Dingo's classes (__repr__()).

**Further steps:**
- I surmise that all mentioned issues regarding non-determinism (grid topology within load areas as well as branch id (#97), number of nodes (#98)) are related to the behaviour described above
- Further investigations should start at [find_nearest_conn_objects()](https://github.com/openego/dingo/blob/5935fd0a03329a9432f52340126878435e657c47/dingo/grid/mv_grid/mv_connect.py#L660)
- To speed up identifying the "salt" a library could be used to track differences between runs (which one?)

Summary: Most wanted Christmas present: A dustpan for all the salt