openego/ding0 #46: Add meta description

gplssm, 2016-07-12:
Add a meta description to auto-generated documentation of Dingo that contains
* Methodological approach
* Use cases
* ...

gplssm, 2017-05-21:
Minimum docs to complete for a first release
- [ ] Welcome page of RTD
- [ ] Readme on GitHub
- [ ] Getting started
  - [ ] Installation instruction (as short as possible)
  - [ ] Provide overview on examples and how to adapt from them

Improvement on API docs is now done in #45 as will probably contained in the next release.

gplssm, 2017-05-31:
Everything (apart from #130 ) is done. Only "What's new" remains. There two options to fill this section

1. Provide a diff report relative to already released pre-version v0.1.0-pre (hard way)
2. Say that 0.1.0 is the first version and everything is new. No further detailing (easy short-cut way)

@nesnoj what's your preference?

nesnoj, 2017-06-01:
2.

closed 2017-06-01 by gplssm