openego/ding0 #49: some satellites are not connected to grid

nesnoj, 2016-07-15:
Example:

[image]

**Reason**

When a satellite is to be connected,
1. a buffer with a predefined [radius](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/config/config_calc#L29) is created around it in function [`calc_geo_branches_in_buffer`](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/tools/geo.py#L6) to determine possible connection branches.
2. Afterwards, a list of possible connection points on these branches is created in [`find_nearest_conn_objects`](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/grid/mv_grid/mv_connect.py#L16), sorted ascending by distance.
3. Then, [`find_connection_point`](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/grid/mv_grid/mv_connect.py#L75) goes through the list from 2. and tries to connect. If a connection point is already within a satellite string (group), it checks whether the satellite can join the group (regarding [maximum load](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/config/config_calc#L17) and [maximum string length](https://github.com/openego/dingo/blob/854cec894c0381f96257c9d951b273385fef80ad/dingo/config/config_calc#L23))
- if no: check next conn. object in list
- if yes: everything is fine

**And now for the crucial stuff:** In most cases, the list generated in 2. contains objects that are located in strings AND objects that are located in ring main lines. Hence, if all groups are "full", the satellite is connected to a (more far away) ring main line. In some cases, the list doesn't contain one of the latter so it WON'T BE CONNECTED.

**Possible solutions:**
1. Memorize the first (nearest) connection object. If the satellite was not connected to any of the objects in the list of 2. (see above), connect it to the first one despite the violation of the two thresholds mentioned above.
2. The initial radius has to be greater to ensure the inclusion of at least one ring main line.
3. Do a second connection process with all nodes that were not connected in the first one. Choose are greater radius.

gplssm, 2016-09-14:
#68

nesnoj, 2016-12-05:
Note: To find isolated nodes, Grid's method `graph_isolated_nodes()`, commited in a9534be, could be useful (e.g. to excluse them from export to PF).

nesnoj, 2016-12-22:
Done in 16bfc8f519bdaa4b8e7ea58b39dcdd74010def27 using solution no. 3 (see above):
- [mv_connect_satellites()](https://github.com/openego/dingo/blob/d9aa33dde49281757f9dc150f0fca3393b1b5ee3/dingo/grid/mv_grid/mv_connect.py#L483) has got 2 modes ('normal' and 'isolated') now and can be controlled using argument `mode` (see docstring for details).
- Possible conn. points for the isolated satellites are determined [using branches of MV rings](https://github.com/openego/dingo/blob/d9aa33dde49281757f9dc150f0fca3393b1b5ee3/dingo/grid/mv_grid/mv_connect.py#L555-L563).
- There's is a separate call of mv_connect_satellites() with `mode='isolated'` in [MVGridDingo.routing()](https://github.com/openego/dingo/blob/d9aa33dde49281757f9dc150f0fca3393b1b5ee3/dingo/core/network/grids.py#L226) to connect the nodes that could not be connected in normal mode.

The result looks a bit messy but now all satellites are member of MV grid:
_Example 1:_
[image]

_Example 2:_
[image]

closed 2016-12-22 by nesnoj