openego/ding0 #59: Choice of LV grid model if apartments exceed 196

gplssm, 2016-07-29:
List of LV grid models ranges from 1 to 196 apartments. What if a LV grid district contains larger size of population resulting in >196 apartments per LV grid district?

If solution is discussed, remove this hack over here https://github.com/openego/dingo/blob/387c2236b1edca49dea95abfd91dd5a097684620/dingo/core/network/grids.py#L312-#L313

gplssm, 2016-12-08:
*Idea if 196 is exceeded*
* add new trafo (smallest size possible) in parallel to existing one
* add branches according to apartments numbers that exceed 196

nesnoj, 2016-12-08:
I agree to point 2 but I propose to check how many LV grid districts are actually affected (have got >196 ap.) prior to the decision on how we deal with no. 1.
Background: In grid reinforcement measures, you probably wouldn't strengthen your grid by installing a small trafo beside a large one but go for one with the same nominal power. But this has to be discussed again..

nesnoj, 2017-04-12:
Result: If count of apartments (Hausanschlüsse) exceeds 196, the residential peak load is distributed equally to these 196.

closed 2017-04-13 by nesnoj