openego/ding0 #76: Lines have length of zero meters

gplssm, 2016-11-28:
Due to some unknown reason, DINGO creates lines with length of zero meters. As this leads to trouble in power flow calculation, those lines are caught by additions of 76e61813a980ce18bd072d0defb0a94ef642f7c1.

TODO
 - solve the original issue of lines with length 0 m
 - remove additions of 76e61813a980ce18bd072d0defb0a94ef642f7c1

nesnoj, 2016-12-05:
I assume this is due to missing geoms of generators. If the `geom_new` field is missing, EnergyMap's `geom` is used and PLOPP, there are a couple of generators at the same position => length of interconnecting line is 0. It will probably be solved by #82...hopefully :).
Nevertheless, a fallback option is useful and should be done during [connection of generators](https://github.com/openego/dingo/blob/a9534be04a18b483e4bfc24592ad8f55d972c9c9/dingo/grid/mv_grid/mv_connect.py#L707) instead of checking it during the export to PF.

nesnoj, 2016-12-09:
Done in 3698803

closed 2016-12-09 by nesnoj