openego/ding0 #81: import of pf results crashes

nesnoj, 2016-12-05:
With grid district id=489, results from PF are imported properly but it crashes with other ids, e.g. id=360:

error from pandas:
> the label [MV_360_lac_64619] is not in the [index]
(seems to be a plotting issue)

nesnoj, 2016-12-06:
right now, with id=489 it crashes too with the same error message..WTH

nesnoj, 2016-12-06:
The problem seems to occur with _load area centres (lac)_ only when importing PF results (see [here](https://github.com/openego/dingo/blob/34750b2f4b63b5e8846d7894f9af54085c89b5f6/dingo/tools/pypsa_io.py#L444)).
1. Reason: there are no _lac_ in the database at all.
2. Original reason: Due to a bug within the export to PF, the check if a load_area is connected to the graph is not done properly (it was called as attribute, not as method). Fixed in 7479356.

The PF now throws a different error:

> ...
>   File "/home/jonathan/git-repos/openego/dingo/dingo/tools/pypsa_io.py", line 395, in run_powerflow
>     network.pf(snapshots)
>   File "/home/matthias/git-repos/PyPSA/pypsa/pf.py", line 98, in network_pf
>     _network_prepare_and_run_pf(network, snapshots, verbose, skip_pre, sub_network_pf, calculate_Y, x_tol=x_tol, use_seed=use_seed)
>   File "/home/matthias/git-repos/PyPSA/pypsa/pf.py", line 57, in _network_prepare_and_run_pf
>     network.determine_network_topology()
>   File "/home/matthias/git-repos/PyPSA/pypsa/components.py", line 791, in determine_network_topology
>     adjacency_matrix = self.adjacency_matrix(passive_branch_types)
>   File "/home/matthias/git-repos/PyPSA/pypsa/graph.py", line 109, in adjacency_matrix
>     shape=(no_buses, no_buses))
>   File "/usr/local/lib/python3.4/dist-packages/scipy/sparse/coo.py", line 184, in __init__
>     self._check()
>   File "/usr/local/lib/python3.4/dist-packages/scipy/sparse/coo.py", line 234, in _check
>     raise ValueError('negative row index found')
> ValueError: negative row index found

Any ideas, @gplssm?

gplssm, 2016-12-07:
This is definitely related to #84 Once, this issue is solved, I'll check for further necessary to fix this issue here.

*Possible reason for this problem:* _if cases_ in the re-import of PF results are differently defined then in the export.

nesnoj, 2016-12-09:
Ok, same problem as stated in the first post with grid district id=359:

> the label [MV_359_lac_21489] is not in the [index]

(whereas it runs properly with grid district id=489)

gplssm, 2016-12-09:
This is solved by e8cce77e0ea7f6c4d9505b8d60310a2b43441f77 and/or a7e1563da6f19d0e029eedb70f8104ec06f35488

closed 2016-12-09 by gplssm