openego/ding0 #84: no stations for LV load areas / grid districts - URGENT

nesnoj, 2016-12-06:
After importing MV Grid Districts, there is a lack of LV stations that are assigned to LV grid districts. Is it still work in progress? (by @jong42)

jong42, 2016-12-06:
The LV stations are in model_draft.ego_grid_lv_onts. Currently there are no LV grid districts and due to  computing time it will be difficult to create them for whole germany.  Maybe I can write a simplified algorithm for generating the LV grid districts

nesnoj, 2016-12-07:
Thank you for your fast reply. Until now I wasn't aware that @gplssm has excluded the LV grid districts temporarily as a workaround. So the reason for not having a country-wide dataset the computing time only or is there more work to be done by you concerning your method?
By the way, about what time are you talking about? days? weeks?

gplssm, 2016-12-07:
The issue is definitely related to the new _data processing_ run. Should be fixed after a re-run of two relevant scripts, that should be completed tomorrow morning.

nesnoj, 2016-12-08:
@jong42, is it possible to fix your bugs and create a new LV grid district dataset by the end of next week?

gplssm, 2016-12-09:
This issue was related to error in _data processing_. Those are solved!

closed 2016-12-09 by gplssm

jong42, 2016-12-09:
Until the end of next week there will be LV grid districts available, but with a few limitations:
- LV grid districts could overlap
- LV grid districts  won't cover the load areas completely, small portions without coverage will likely occur
- Very large load areas won't have neither LV substations nor LV grid districts due to a problem in the substation creation

Can you work with these limitations? If not,  the algorithm has to be changed again

nesnoj, 2016-12-12:
Since the original topic is different, I moved the conversation about a new LV grid district dataset here: #93