openego/ding0 #88: LV generator allocation to LV grid districts

nesnoj, 2016-12-08:
Currently, the LV generators are allocated to LV load areas (method implemented by @Ludee in [sql snippets ego_dea_allocation*](https://github.com/openego/data_processing/tree/master/dataprocessing/sql_snippets)) and have to be allocated to LV grid districts afterwards.

Possible criterions for this distribution could be:
1. LV Grid district's area
2. Count of apartments (proportional to roof area, valid in regions where house types are similar)
3. ...

This depends on a complete grid district dataset (#84).

nesnoj, 2016-12-22:
Oddly enough the column in table `model_draft.ego_dea_allocation` which is supposed to contain the id of LV load area (for voltage levels 06+07) is empty.
Any hint on that @Ludee?

jong42, 2016-12-22:
Since the LV load areas were created last week, maybe the generator table isn't updated yet?

nesnoj, 2016-12-22:
Yea, the DB is currently updated in order to switch to v0.2, @Ludee wrote that it will be finished by tonight..

jong42, 2016-12-22:
In case that an outdated table name is the cause: The current LV grid districts are in model_draft.ego_grid_lv_griddistricts_temp

closed 2017-04-11 by nesnoj