openego/ding0 #91: use DataFrames instead of database to interface pypsa

nesnoj, 2016-12-09:
Profiling reveals that the export of MV grid to db tables for pypsa actually takes some time:

[image]
We better go for pandas DataFrames to pass the data to pypsa.
We keep the database structure as possibility for exchanging grids.

gplssm, 2016-12-09:
Indeed!

I would suggest to take (and maybe adopt) current database structure for grid to export the final result.

gplssm, 2016-12-10:
See progress in #92

Still pending tasks
- [ ] pq_set and v_mag_set via DataFrame
- [ ] `run_powerflow()` based on DataFrames

gplssm, 2016-12-13:
Done. Discovered issues with new spicy salt that gets involved when generators are connected. See over here #98 

Performance
 - via DataFrames: ~16s (new)
 - via database: ~46s (before)

closed 2016-12-13 by gplssm