openego/ding0 #93: new LV grid district dataset

nesnoj, 2016-12-12:
We need a new country-wide dataset for LV grid districts. The discussion came along the way in #84 and I moved it here since it deserves an own issue.

@jong42 wrote:
> Until the end of next week there will be LV grid districts available, but with a few limitations:
> - LV grid districts could overlap
> - LV grid districts  won't cover the load areas completely, small portions without coverage will likely occur
> - Very large load areas won't have neither LV substations nor LV grid districts due to a problem in the substation creation
> 
> Can you work with these limitations? If not,  the algorithm has to be changed again

It'd be great if you could finish this makeshift dataset by the end of this week and - if possible - produce a final one by the end of 2016.

jong42, 2016-12-16:
a temporary dataset is available as model_draft.ego_grid_lv_griddistricts_temp. These are simply circle-shaped buffers around the substations. Hopefully the generation of voronoi-Polygons can be improved and used to replace the temporary dataset next week

nesnoj, 2017-02-22:
Dataset works even though not perfectly.
Further discussions take place in https://github.com/openego/data_processing/issues/47

closed 2017-02-22 by nesnoj