openego/ding0 #94: Cross-check assumptions regarding power flow calculation in MV grid

gplssm, 2016-12-12:
Currently PF is configured as follows

Load case: P=peak load, q=via cos phi; no generation. Cross-check active/reactive power calculation here

Feed-in  case: P=nominal capacity, Q=0 ; no load

nesnoj, 2017-01-13:
We use assumptions according to dena VNS (load/generation: p.98, cos phi: p.114)

nesnoj, 2017-05-23:
According to DSO our current approach (load case: max. load, no generation, feedin case: max. generation, no load) is valid. dena VNS uses technology-specific values (p. 98).
@gplssm Do you think we should adjust our code to allow the dena version?
If so, we could pimp the [`control_generators()`](https://github.com/openego/dingo/blob/f3e35adff469443bfec8473c1bc8abda5e1ad4cb/dingo/core/network/__init__.py#L309) method and pass tech-specific factors to be applied recursively.

nesnoj, 2017-05-29:
Moved to nice-to-have.