openego/ding0 #98: Salted number of node/buses

gplssm, 2016-12-13:
Analyzing number of buses revealed that grid topology varies from model run to model run. If we look at number of nodes in the graph, isolated node and resulting number of buses, we see that either (using MV grid 489)

1. In total there are 174 nodes and 168 buses
2. In total there are 176 nodes and 170 buses

In both cases number of isolated nodes is the same: 6.

The missing/additional two nodes are mv_cable_dist_48 and mv_cable_dist_49.

Disabling connection of generators results in 167 nodes, 19 isolated ones and thereof 148 buses result. Within 20 runs, this result didn't change. Whereas alterations with generators included occur every second to fifth run. Thus, the actual issue probably lies in the connection of generators

nesnoj, 2016-12-20:
Actually I cannot confirm that this behaviour is related to the generators only. It is not related to the generators' order itself because it is sorted prior to connection process. It is rather related to grid's topology before. I discovered some more salt that is spilled when connecting stations and satellites..
I'm on it..

nesnoj, 2016-12-20:
I dived quite deep into DINGO's salt but I have to leave this issue unresolved for now due to lack of time. The current status is described in #40, resume discussion about determination over there.

nesnoj, 2017-08-22:
I guess this one was solved in #200 (5.)

closed 2017-08-22 by nesnoj