openego/ding0 #101: Make data file access failsafe

gplssm, 2016-12-19:
Currently, when executing `test_dingo2.py` from arbitrary location, script fails on reading data files (.csv).

```
(dingo_release) guido@rli-nb-17 ~/rli_local/ego.io $ python3 ../dingo/test_dingo2.py 
Traceback (most recent call last):
  File "/home/guido/rli_local/dingo/dingo/core/__init__.py", line 295, in import_mv_grid_districts
    lv_stations)
  File "/home/guido/rli_local/dingo/dingo/core/__init__.py", line 345, in import_lv_load_areas
    'equipment-parameters_LV_cables.csv'), index_col='name')
  File "/home/guido/virtualenvs/dingo_release/lib/python3.4/site-packages/pandas/io/parsers.py", line 645, in parser_f
    return _read(filepath_or_buffer, kwds)
  File "/home/guido/virtualenvs/dingo_release/lib/python3.4/site-packages/pandas/io/parsers.py", line 388, in _read
    parser = TextFileReader(filepath_or_buffer, **kwds)
  File "/home/guido/virtualenvs/dingo_release/lib/python3.4/site-packages/pandas/io/parsers.py", line 729, in __init__
    self._make_engine(self.engine)
  File "/home/guido/virtualenvs/dingo_release/lib/python3.4/site-packages/pandas/io/parsers.py", line 922, in _make_engine
    self._engine = CParserWrapper(self.f, **self.options)
  File "/home/guido/virtualenvs/dingo_release/lib/python3.4/site-packages/pandas/io/parsers.py", line 1389, in __init__
    self._reader = _parser.TextReader(src, **kwds)
  File "pandas/parser.pyx", line 373, in pandas.parser.TextReader.__cinit__ (pandas/parser.c:4175)
  File "pandas/parser.pyx", line 667, in pandas.parser.TextReader._setup_parser_source (pandas/parser.c:8440)
FileNotFoundError: File b'dingo/data/equipment-parameters_LV_cables.csv' does not exist

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "../dingo/test_dingo2.py", line 30, in <module>
    nd.import_mv_grid_districts(conn, mv_grid_districts)
  File "/home/guido/rli_local/dingo/dingo/core/__init__.py", line 300, in import_mv_grid_districts
    raise ValueError('unexpected error while initiating MV grid_districts from DB dataset.')
ValueError: unexpected error while initiating MV grid_districts from DB dataset.
```

Two issues are to be solved here!

1. Change file access to use absolute paths. For inspiration take a look [here](https://github.com/oemof/oemof/blob/6c451c4c7f940b28efb8fa8feb9a139345f483de/examples/solph/simple_dispatch/simple_dispatch.py#L37-#L40)
2. To enable proper execution in virtualenv, include data files in installation process as done [here](https://github.com/oemof/oemof/blob/6c451c4c7f940b28efb8fa8feb9a139345f483de/setup.py#L23-#L38)

nesnoj, 2016-12-23:
2. was done in b3ebb5d35cdb990eab59b60a30867d82a0f87387

nesnoj, 2017-01-15:
1. was done in b84b463eeef95bd7271110180c496fa4022bd4ec using package path.

closed 2017-01-15 by nesnoj