openego/ding0 #102: PF crashes for some MV grid districts

nesnoj, 2016-12-21:
PF crashes with MV grid district id = 359 stating

> Traceback (most recent call last):
>   File "/home/jonathan/git-repos/openego/dingo/test_dingo2.py", line 52, in <module>
>     mv_grid_district.mv_grid.run_powerflow(conn, method='onthefly')
>   File "/home/jonathan/git-repos/openego/dingo/dingo/core/network/grids.py", line 548, in run_powerflow
>     pypsa_io.run_powerflow_onthefly(components, components_data, self)
>   File "/home/jonathan/git-repos/openego/dingo/dingo/tools/pypsa_io.py", line 708, in run_powerflow_onthefly
>     network.pf(snapshots)
>   File "/home/matthias/git-repos/PyPSA/pypsa/pf.py", line 98, in network_pf
>     _network_prepare_and_run_pf(network, snapshots, verbose, skip_pre, sub_network_pf, calculate_Y, x_tol=x_tol, use_seed=use_seed)
>   File "/home/matthias/git-repos/PyPSA/pypsa/pf.py", line 57, in _network_prepare_and_run_pf
>     network.determine_network_topology()
>   File "/home/matthias/git-repos/PyPSA/pypsa/components.py", line 791, in determine_network_topology
>     adjacency_matrix = self.adjacency_matrix(passive_branch_types)
>   File "/home/matthias/git-repos/PyPSA/pypsa/graph.py", line 109, in adjacency_matrix
>     shape=(no_buses, no_buses))
>   File "/usr/local/lib/python3.4/dist-packages/scipy/sparse/coo.py", line 184, in __init__
>     self._check()
>   File "/usr/local/lib/python3.4/dist-packages/scipy/sparse/coo.py", line 236, in _check
>     raise ValueError('negative column index found')
> ValueError: negative column index found

With id=480 or 489 it runs properly. The same error already came up in the past (#81).

gplssm, 2016-12-22:
First, I inspected length of DataFrames contained in `component` against `component_data`.

Results
* 396 buses, 396 `v_mag_set` data :+1: 
* 396 generators, 60 `p` and `q` data :-1: 
* 226 loads, 226 `p` and `q` data :+1: 

396 buses and 396 generators seems to be suspicious! I'll dig deeper into that!

nesnoj, 2016-12-22:
I guess u got lost in the ground?

nesnoj, 2016-12-23:
I did some testing:
- The PF seems to crash for MV grid districts that contain at least one LV load area of type aggregated (e.g. grid district ids = 359, 415, 424, 447)
- It runs properly for grid districts without aggregated LV load areas (e.g. grid district ids = 402, 406, 489, 480, 371)

To used the following snipped to test if there's an aggregated load are within MV grid district (see 2fbc5a820a865305fe0afebaae570ea88b520564):
```
if any([_.is_aggregated for _ in nd._mv_grid_districts[0].lv_load_areas()]):
    print('GRID DISTRICT CONTAINS AGGREGATED LV LOAD AREA')
else:
    print('GRID DISTRICT DOES NOT CONTAIN AGGREGATED LV LOAD AREA')
```
So this bug may be related to aggregated LV LA...

gplssm, 2016-12-23:
I was diggin' around as well, but I reached just middle of the earth... ;)

Great hint about the aggregated load areas!

gplssm, 2016-12-23:
I solved the issue in by excluding `LVLoadAreaCentreDingo` from
1. export of lines: no line will be exported when one of the connected nodes is a `LVLoadA...`
2. import of bus results: no data is imported, when type of node `LVLoadA...`

I would recommend to move the creation of lists of relevant nodes/edges (refer code changes in 2417b7865ca76600aa2c18ba842ba7433499c2e2) to a central place. Maybe as attribute to the grid. As long as this isn't done, I wouldn't close this issue.

nesnoj, 2016-12-23:
Thanks for the fix!
If I get this properly you mean a list of nodes the PF has to be performed for?

BTW: U did your commits in features/export2pypsa but I merged to dev before. So I merged again - any further modifications are to be done in dev plz :)

nesnoj, 2016-12-23:
Ok, I'll move that to the 0.1 Release milestone

gplssm, 2016-12-23:
Yeah, you can postpone that.

The actual problem is, that we connect `LVLoadAreaCenters` as load directly connected to the busbar of the substation. Compared to handling of all other types of nodes this is odd. It necessitates a special handling of nodes and edges lists used in the PyPSA-export/importer, such that

* No bus is created for `LVLoadAreaCenters` during export
* No bus of `LVLoadAreaCenters` is expected during import of results
* No line is created between an arbitrary node and `LVLoadAreaCenters`

You can see which lists I'm talking about in the changes I did in 2417b7865ca76600aa2c18ba842ba7433499c2e2

nesnoj, 2017-01-10:
Found another bug: PF crashes with grids which have got no generators - fixed in 09f0875aa3d976d7a218db6cec1af7bf7b3ae7cf

nesnoj, 2017-04-11:
I think this one is solved, @gplssm?
If so, plz close

nesnoj, 2017-05-10:
@gplssm plz close if done

gplssm, 2017-05-16:
I think we can close it. No problems occurred in the past months..

closed 2017-05-16 by gplssm