openego/ding0 #103: LV trafo types for very small LV grid districts

nesnoj, 2016-12-22:
The smallest LV trafo is 100 kVA - way too large for very small LV grid districts. We may introduce 30 and 50 kVA types (definition is done in [relation_apartment_trafo.csv](https://github.com/openego/dingo/blob/0ec3f0a86ad65aec2b2a0edeb69aaae5d106b667/dingo/data/relation_apartment_trafo.csv))

nesnoj, 2017-05-09:
Couldn't find sources with R+X - may get from Sincal..

nesnoj, 2017-07-24:
I added a 50 kVA type using @jochenbuehler's method in 376a2170cb696174f1d1d319a6b9b1ef34ce32d2:
`R = u_Rr/100% * U_US^2/S` and `X = u_kr/100% * U_US^2/S` using u_Rr=2.45 and u_kr=6.15

closed 2017-07-24 by nesnoj

gplssm, 2017-07-25:
:rocket:

jochenbuehler, 2017-07-25:
We just shouldnt Forget to mention in the final report that the data of the 50kVA transformer is not includet in the data of our source (Torsten Werth; Netzberechnung mit Erzeugungsprofilen)