openego/ding0 #108: isolated LV load areas

nesnoj, 2017-01-12:
Some LV load areas got no grid although they contain ONS and centre point - example (mvgd id=480, lvla id=63250):

[image]

nesnoj, 2017-01-12:
There's no entry for lvla id=63250 in the peak load table - currently the table calc_ego_loads.calc_ego_peak_load is used but may be outdated?
Is the calculation of peak loads part of data proc?

Ludee, 2017-01-15:
Yesm [peak_load_per_load_area.py](https://github.com/openego/data_processing/blob/master/dataprocessing/python_scripts/peak_load_per_load_area.py) is part of eGoDP.
@gplssm is author. Will have a look

closed 2017-04-11 by nesnoj