openego/ding0 #111: allocation of LV generators

nesnoj, 2017-01-25:
In the current methodology of generator allocation, LV RES are randomly distributed on all points on a 50x50 m lattice created within LV load areas, the following is to be discussed:

1. Since the population count is not considered, it may happen that there're too many RES allocated to a LV LA - this is to be examined.
2. There's no distinction between aggregated and non-aggregated LV LA in the status quo scenario, hence assuming an equal distribution (random) over all points (count is proportional to LA LV area) a major count of RES will be probably allocated to agg. LV LA. This might be inconsistent with the argumentation "agg. LV LA do not contain a significant amount of RES". At least, for the future scenarios (done by @wolfbunke) this allocation method is to be re-evaluated.

wolfbunke, 2017-01-30:
The allocation method is not finished yet due to work shops, holidays and many more. I am looking forward to finish this task this week by implementing @Ludee methods for all units.

Methods: https://github.com/openego/data_processing/tree/refactor/oedb-restructuring_v0.2/dataprocessing/renewable_energy_allocation

Ludee, 2017-01-30:
The rea methods were debugged last week and successfully executed on Friday. So pull the v0.2 branch and call me if you have problems with it.

gplssm, 2017-04-13:
@nesnoj  Isn't this solved now?

nesnoj, 2017-04-13:
No.
It's related to #123. If we modify initial RES allocation, we have to consider this for future scenarios too.
@wolfbunke I'd like to discuss this issue with you as soon as you're back at work.

wolfbunke, 2017-05-02:
Hey @nesnoj I am back in office. Your new phone number is not working or?

gplssm, 2017-05-02:
Our move to the new office isn't completed yet, maybe telephone isn't working... We're back at the office tomorrow. @nesnoj will call you.

nesnoj, 2017-05-03:
Currently our telephone services are still offline due to our move to Adlershof. I'll come back to you within the next days..

gplssm, 2017-05-23:
@wolfbunke creates a new table with RES power plants for 2035, 2050 and drops a line.
Then @gplssm analyzes following potential issues

1. The REA-methods applied on future RES power plant capacities increases the issue of power plants that can't be allocated to appropriate areas
2. Are too many RES power plants re-located to aggregated load areas?
3. What are maximum line length the occur due to non-re-located power plants?
4. Start to create a longer report on RES location analysis and present on next project meeting. Maybe hand-over to Sherif

mltja, 2023-03-13:
Deprecated to new data, new method in #380

closed 2023-03-13 by mltja