openego/ding0 #116: treating special MVGDs

nesnoj, 2017-02-17:
There're some odd MVGD:

MVGD...
1. ...is completely empty - no LVLA, no genos (e.g. id=1328)
2. ...only containes aggregated LVLA (e.g. id=1294)
3. more?

1.+2. were temporarily handled in 03964c752126a6c6f52bf4bc13fc8914355ee5ef introducing a new method of NetworkDingo (they are deleted from Network).

ToDOs:
- concerning 1: Discuss with transmission grid
- concerning 2: Make sure the load from agg. LVLA is assigned correctly to HV/MV station, so deletion might be the wrong way (instead an attribute 'is_valid' could be used with an optional reason attached..)

nesnoj, 2017-02-23:
See related issue in dp: https://github.com/openego/data_processing/issues/54

piaulous, 2023-03-13:
Deprecated, new data @openego/egon-data  / methodology in #380.