openego/ding0 #118: Removing invalid MV grid district fails on certain MVGD

gplssm, 2017-03-26:
`nd.validate_grid_districts()` should remove MV grid districts from list for a set of reasons. This works sometimes, sometimes not.

For example:

a) Run Dingo for a single MVGD, in this case with id 2, we get this message `MV Grid District mv_grid_district_2 contains only aggregated LV Load Areas and was removed`. Dingo stops. Everything is fine.
b) Passing a list of MVGDs to Dingo (i.e. list(range(1,5)) ),  `nd.validate_grid_districts()` does the following
```
Original list: [mv_grid_district_1, mv_grid_district_2, mv_grid_district_3, mv_grid_district_4]
After validation: [mv_grid_district_2, mv_grid_district_4]
```

Would be nice to get this function working properly in order to run multiple MVGDs in a single run while excluding invalid MVGDs via this function.

nesnoj, 2017-03-26:
not sure if i got this right - in your example b) mvgd 1+3 were removed although they are not faulty?

gplssm, 2017-03-26:
Almost. 1+3 are removed, which is fine. But 2 should be removed as well. For some unknown reason it's not.

nesnoj, 2017-03-26:
example list?

gplssm, 2017-03-26:
Try

a) mv_grid_districts = [2]
b) mv_grid_districts = [1,2,3,4]

Case a) should abort appropriately, case should crash on mvgd 2.

nesnoj, 2017-03-26:
The reason is quite simple: The mvgd generator `nd.mv_grid_districts()` is used in the loop but altered within the loop at the same time!

(over and out)

commit 5ae5979c (2017-03-26): fix #118
closed 2017-03-26 by nesnoj