openego/ding0 #127: Consider retail/industrial/agricultural in LV grids

gplssm, 2017-04-12:
Two cases are differentiated

1. retail+industrial peak load <=300 kVA: an additional branch is added to Kerber-based grid topology
2. retail+industrial peak load >300 kVA: the loads are connected to an individual LV grid including an additional transformer which is located in the existing substation

- [ ] Put grid data into repo in appropriate format
- [ ] Write method to construct the additional branch/additional grid within Dingo

nesnoj, 2017-04-12:
We use LV grids from [1] (using profile G0 for each load):
[image]

For 1. we use a single branch line from each of the two grids, for 2. the whole grid.
To be determined: Which load do we assume per connection point?
How to we account for reactive power? In [1], cos_phi=0,85 is assumed (todo: compare to dena).

[1] Schaller et al.: "Modellierung realitätsnaher zukünftiger Referenznetze im Verteilnetzsektor zur Überprüfung der Elektroenergiequalität"

gplssm, 2017-04-27:
The above proposed grid model is actually a model for MV grids.

A new idea for representing retail/industrial and agricultural demand is described in the following.

Retail/industrial and agricultural demands are connected to the grid via separate branches for (a) retail/industrial and (b) agricultural.

Based on the maximum length of branch lines of <= 400 m of retail grids [Bodach, 2006], we assume that loads (retail/industrial) connected to this branch lines are on average in the mid of the branch line.
In case of agricultural demand, the maximum length of branch lines adds up to 800 m.

Single retail/industrial and agricultural demands a equally distributed on this branch lines. The of individual demand connected, determines of the areas for each sector identified for the LVGD.

The number of branches considered for retail/industrial and agricultural defines of the total sectoral peak load in each LVGD. Up to 300 kVA are connected to a single branch line [Dena VNS]. Thus, the number if branch lines determines as `ceil(total_load / 300)`

gplssm, 2017-05-02:
For progress see #159

gplssm, 2017-05-02:
Grid data does need to be saved in CSV or similar. Just a few assumptions added to config_calc are sufficient to construct the grid topology.

Fundamental work is done, a method is added that constructs LV grid branches for sectors retail/industrial and agricultural.
Remaining task are described in #77 #128 #147 #154 #157 #158 
When these completed, we merge #159

commit 0cb506e6 (2017-05-02): Add method build_lv_graph_ria()

Builds LV graph structure for sectors retail/industrial and agricultural

Close #127
closed 2017-05-02 by gplssm