openego/ding0 #129: Allocation of generator (REA) to house connections

gplssm, 2017-04-12:
We distinguish 3 sizes of generators (all types of generators)

1. \>100kW: Based on location provided by _data processing_ these are connected to secondary side of MV-LV transformer
2. \>=30 kW ... <=100 kW: preferably connected to retail/industrial or agricultural branch or grid with minimum distance grid connection point. If such a grid is not available, connected to _normal_ grid.
3. <30kW: These remaining generators are assumed to be located on roofs of residential buildings. Allocation method is described below.

The allocation of generators (identified to be located in a LV grid district) to house connection is based on _fixed random_. This means, random distribution is done once and fixed. This distribution is then given per LV grid type (1..196) and will be applied for each LVGD where this particular LV grid type is used. Generators within this LVGD are allocated according to this distribution.
The distribution is hold in a data structure of IDs _(i,j)_, where _i_ represents type of branch and _j_ represents count of house connection start counting from the MV-LV substation.

Open questions
- What if number of generators exceed number of house connections?

nesnoj, 2017-05-07:
Answers:
1. New random series are created until all genos are assigned.
2. Connect genos to MV-LV substation

nesnoj, 2017-05-08:
Current state of implemented method including methodical changes in function [`lv_connect_generators()`](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L11):

1. [\>100kW (v_level = 6)](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L47): Connect to LV station (real distance is calculated).
2. [\>30 kW ... <=100 kW (v_level = 7)](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L78): Random allocation to LV load of type retail, industrial, agricultural. Connect to cable dist. within building.
3. [<=30 kW (v_level = 7)](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L63): Random allocation to LV load of type residential. Connect to cable dist. within building.

Further assumptions/notes:
- For 2.+3.: Length = 1m
- For 2.+3.: If no appropriate LV load exists, generator is [connected to station](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L92).

Furthermore, the drawing method [GridDingo.graph_draw()](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/core/network/__init__.py#L67) now can plot  LV grids in a hierarchical-like structure.

ToDos:
- [ ] Check resulting grids (using the drawing method, it is temporarily enabled [here](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/core/__init__.py#L1245)). Closing the first plot brings up the second...
- [ ] Set type of branches ([here](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L52) and [here](https://github.com/openego/dingo/blob/cb91bd920ecce9d53b1d0f9e305211185537826a/dingo/grid/lv_grid/lv_connect.py#L104))

gplssm, 2017-05-09:
Ideas for above TODOs

a. Print stats a the end of example. Maybe put code for this into some subdirectory/separate file an call this from example
b. Enable plotting via debug flag. Check some exemplary grids
c. Type of branch is chosen according to nominal power of generator, similar to sectoral branch lines

gplssm, 2017-05-09:
Plots of LV grid topology are really nice. Great stuff Mr. Johnson :+1: 
Haven't checked the content in detail, but looking at it roughly it looks plausibel. It put plot creation behind a debug flag provided to `nd.connect_generators(debug=False)`.

gplssm, 2017-05-09:
Generators that are directly connected to MV-LV substations are printed now. Introduced by bfb888c90cf66776df6d49fea7114af00f64723a

Analyzing MVGD 3545, no generator is connected directly to the bus bar.

nesnoj, 2017-05-09:
> Generators that are directly connected to MV-LV substations are printed now. Introduced by bfb888c Analyzing MVGD 3545, no generator is connected directly to the bus bar.

That's kind of weird. While debugging MVGD 3545 I came across some LVGD with genos but no loads - they are connected to MV-LV station.. I get a lot of corresponding [WARNINGS](https://github.com/openego/dingo/blob/bfb888c90cf66776df6d49fea7114af00f64723a/dingo/grid/lv_grid/lv_connect.py#L99-L104) -> check your log. May I got u wrong?

gplssm, 2017-05-10:
True, that's what the log was intended for... ;-)

The lines of code I wrote to get all generators directly connected to the bus bar of the station finds no generator because generator list of LV grids are empty.
I think generators should be added to the grid via `lv_grid.add_generator()` in [lv_connect_generators()](https://github.com/openego/dingo/blob/bfb888c90cf66776df6d49fea7114af00f64723a/dingo/grid/lv_grid/lv_connect.py#L12), right?

gplssm, 2017-05-10:
@EdithaK decided fallback method is good enough for now

nesnoj, 2017-05-10:
> I think generators should be added to the grid via lv_grid.add_generator() in lv_connect_generators(), right?

Nope, they are already added during import, see [here](https://github.com/openego/dingo/blob/bfb888c90cf66776df6d49fea7114af00f64723a/dingo/core/__init__.py#L688) (otherwise [this one](https://github.com/openego/dingo/blob/bfb888c90cf66776df6d49fea7114af00f64723a/dingo/grid/lv_grid/lv_connect.py#L45) wouldn't give any results :).

nesnoj, 2017-05-15:
I close this one, validation in #169 
ok, @gplssm ?

closed 2017-05-15 by nesnoj