openego/ding0 #130: LV grid documentation

gplssm, 2017-04-12:
Write a documention in plain english that provides the overview

gplssm, 2017-05-11:
I did some restructuring and added some introductionary addings... 

I reworked existing text until [here](https://github.com/openego/dingo/blob/5da2ad80f381a303891421ee0a924cf185a4744e/doc/about.rst#L1-#L30).

Following TODOs remain

- [ ] Finish [section about residential LV grid branches](https://github.com/openego/dingo/blob/5da2ad80f381a303891421ee0a924cf185a4744e/doc/about.rst#L52-#L96) @EdithaK 
- [ ] Finish [section about retail/industrial and agricultural LV grid branches](https://github.com/openego/dingo/blob/5da2ad80f381a303891421ee0a924cf185a4744e/doc/about.rst#L98-#L101) @gplssm

gplssm, 2017-05-11:
I added a description for the branches retail/industrial/agricultural. If you need, take inspiration how to use tables, equations, figures, and references to literature in sphinx docs.

gplssm, 2017-05-21:
I'm done so far with a first version. @EdithaK you can close the issue when residential section is complete

piaulous, 2023-03-13:
Deprecated due to new LV grid generation method, see #380.
Already mentioned #383.

closed 2023-03-13 by piaulous