openego/ding0 #133: LV grid reinforcement

gplssm, 2017-04-13:
Reinforce LV grid topology after loads and generators are connected

- [ ] Define methodology
- [ ] Implement in Dingo

nesnoj, 2017-05-16:
Please use [ND's main method](https://github.com/openego/dingo/blob/6d248bd3364c05496b243aaf2581a822969c1e7f/dingo/core/__init__.py#L1519-L1522) to call reinforcement.

gplssm, 2017-05-21:
A bunch of code has been created. Work is done! But there are still things to deal with... see below.

Things to cross-check/answer/implement
- [ ] analyzing a few grids, over-voltage up to 10% occured only in load case
- [ ] Impedance seems to be quite high: 30 ... 80 ohm (R and X)
- [ ] There is a high chance voltage issues or overloading remains for some branches due to `s_max` exceeding capacity of largest cable
- [ ] (n-1)-criterion is not considered for MV-LV transformers: cross-check with literature
- [ ] Just a note: Extension of transformers when voltage issues at MV-LV (secondary side) bus bar occur has no effect on voltage level. May voltage issue than is related to impedance of MV grid

gplssm, 2017-05-21:
Close in favor of #169

closed 2017-05-21 by gplssm

nesnoj, 2017-05-22:
Cheers dude! :+1: