openego/ding0 #134: max. voltage drop / overvoltage in MV grid

nesnoj, 2017-04-13:
In the SciGRID paper a voltage level of 104% * U_N and max. voltage drop / overvoltage of 4% is used. This results in a max. voltage drop / overvoltage of 2% in LV part (MV-LV substation and LV grid) which is unrealistic. Use 104% and 2% for future runs.
(DSO use 102.5% ± 1.5% => 104% in worst case)

nesnoj, 2017-05-17:
Also, max. voltage drop is to be respected [during ring creation](https://github.com/openego/dingo/blob/751efc1e3607073e2b146d717e53d79dfedcde46/dingo/grid/mv_grid/models/models.py#L204).

nesnoj, 2017-05-30:
We gotta talk about this one again @gplssm @jochenbuehler

nesnoj, 2017-05-30:
Todos:
- [ ] Use separate U (U_N?) and delta_U for load case (ring creation) and feedin case (reinforcement)
- [ ] Reinforcement: Use U_N instead of U

closed 2017-06-01 by nesnoj

gplssm, 2017-06-01:
I go for a run on the server

commit 0b54a9dd (2017-06-01): fix #134