openego/ding0 #149: consider multiple trafos in one station in PF

nesnoj, 2017-04-19:
In method [select_transformer()](https://github.com/openego/dingo/blob/a275f239d0a45cf89a3005bc55a7a6596d94825d/dingo/core/network/grids.py#L699) multiple trafos are placed according to peak load. This is to be considered in pypsa_io.

nesnoj, 2017-05-17:
I move it to LV grid milestone for now and we move it to eDisGo later.
Related to #85.

gplssm, 2017-05-21:
LV station's transformers are considered by overloading and over-voltage check with parallel operation. This means the have a bus at primary and secondary side connecting all transformers contained in a substation. Nominal apparent power of is considered by aggregated individual nominal power of all transformers

closed 2017-05-21 by gplssm