openego/ding0 #155: LA with no LVGD

nesnoj, 2017-04-28:
There are LA with no LVGD => create one artificial LVGD (geom from LA).

nesnoj, 2017-05-03:
Cannot reproduce.
Inserted catch for this case in e36c11d33313e8322e717eef3f7ed1b04ed86d60 (exception will be raised). Re-open issue again and fix bug if exception is raised..

nesnoj, 2017-10-24:
I ran into this one again in MVGD 2711 (dp v0.3 model draft).
There're 53 LAs without a LVGD:
```
SELECT la.id,
       subst_id,
       zensus_sum,
       round(area_ha, 1) as area_ha,
       round(COALESCE(sector_peakload_retail + sector_peakload_residential + sector_peakload_industrial, sector_peakload_agricultural) * 1000000) as peakload
  FROM model_draft.ego_demand_loadarea as la
  WHERE NOT EXISTS (SELECT lvgd.id
		      FROM model_draft.ego_grid_lv_griddistrict as lvgd
		      WHERE la.id = lvgd.la_id)
  ORDER BY area_ha DESC;
```
As far as I can see, only LAs which were cut by the MVGD boundary are affected (see also https://github.com/openego/data_processing/issues/68).

What does it mean for dingo?
1. LA is aggregated -> no problem (to be fixed once MV grids within agg. LAs are modelled)
2. LA is very small -> no problem (LA can be used as an "artificial" LVGD)
3. Somewhere in between 1. and 2. -> **falsifies MV grid topology**

As this bug affects only 0,025% of LAs it may be ok not to fix it right now, but as it affects 52 different MVGD, this should be fixed one day..

I added an "artificial" LVGD (created from LA data) as temporary solution.

nesnoj, 2017-10-24:
Addition @gplssm:
Consequently, the count of invalid MVGDs should be reduced by 52 by now..

mltja, 2023-03-13:
Already fixed in #380

closed 2023-03-13 by mltja