openego/ding0 #166: create logo

nesnoj, 2017-05-10:
oh yea! I will do this on Friday..

nesnoj, 2017-05-28:
Some drafts, please gimme a vote @gplssm @EdithaK @Ludee @jochenbuehler 

**LOGO A1**
[image]
___
**LOGO A2**
[image]
___
**LOGO A3**
[image]
___
**LOGO A4**
[image]
___
**LOGO A5**
[image]
___
**LOGO B1 (mr. psychodingo)**
[image]
___
**LOGO C1 (o'reilly sytle)**
[image]
___
**LOGO D1**
[image]
___
**LOGO E1**
[image]
___
**LOGO F1**
[image]
___
**LOGO G1**
[image]

nesnoj, 2017-05-28:
Edit: Or add another version :)

gplssm, 2017-05-29:
I'm impressed! I think there is not need to make additional proposals...

My top three are (descending by preference)
C1
E1
A5

jochenbuehler, 2017-05-29:
A5

EdithaK, 2017-05-29:
A3 G1

EdithaK, 2017-05-29:
Stimmt, A5 ist auch gut.

Ludee, 2017-05-29:
Veto on A3 & A5 because of icon in the text! Don't mix.
The animial is a good idea but perhaps only an "insider".

G1 is simple and good.
F1 is favourite because of similarities with the open_eGo logo!
Mix of G1 & F1 would be perfect and match with eGo.

nesnoj, 2017-06-06:
Ok, let's go for **A5**
[image]
Thank you for voting!

U can find the logo in project folder.

closed 2017-06-06 by nesnoj