openego/ding0 #171: Some sections of cfg_dingo are not available after pickle re-import

gplssm, 2017-05-17:
When re-importing `NetworkDingo` that was earlier pickled to a file, some sections of `cfg_dingo` aren't available anymore.

Add 

```
import pickle
pickle.dump(nd,
            open("dingo_grids_{0}.pkl".format(mv_grid_districts[0]), "wb"))
```

then re-import this data with

```
import pickle

a = pickle.load(open('dingo_grids_3545.pkl', 'rb'))
a.to_dataframe()
```

`a.to_dataframe()` runs into to a KeyError because section 'geo' is not available in `cfg_dingo`.

gplssm, 2017-05-17:
The import of `cfg_dingo` takes place in the example.py script and not in the `nd`-object nor in its associated classes. Probably only the config file `config_db_table.cfg` is imported somewhere in `nd`. Thus, its sections are contained in `cfg_dingo`.

Potential solutions
1. Replace `cfg_dingo.get(...)` by hard-coded values or values retrieved from other sources
2. Move `cfg_dingo` imports to the beginning of `nd`

What do you think about option 2, @nesnoj ?

nesnoj, 2017-05-18:
:+1: Option 2 :+1:
U gonna do this?

nesnoj, 2017-05-19:
"Some" measures were required to fix this one:
1. Moved import of config to ND (90c0c1a601b44f8f7f9b3a3d1e3a84493eb46200)
2. Moved import of ORM to ND (bf61b612b05830df6108092e79262ab3533b9d76)
3. Introducing property decorator "network" to all classes for accessing config of ND (7ef7d7779447e17fecf0b3661d2148e52ecd18c0, 1e044f2e7690328ef51e6d085ebbbf5d63d351c8)

So the config is now accessible at `NetworkDingo.config`, from classes `class.network.config`.

Within dingo, the access on config data via cfg_dingo.get() remains as it is, modifications for access via ND's config were implemented in an [extra branch](08f1092c4403af0dfe00b7e11bee4eb44816ce38) but too shabby to use it in dev (for now: we better keep it trapped there).

nesnoj, 2019-01-09:
Concerning the content in the mentioned branch [`feature/get_config_data_from_nd`](https://github.com/openego/ding0/tree/feature/get_config_data_from_nd):
Instead of doing this by hook or by crook using a Network instance, this might be done using @classmethod decorator..

piaulous, 2023-03-13:
Closed, handling of pickle option will be discussed in #386.

closed 2023-03-13 by piaulous