openego/ding0 #177: Harmonize ~/.dingo directory creation in config/results

gplssm, 2017-05-29:
At several locations in the code it is checked for if the ~/.dingo was already created - and is created if not. See for example [here](https://github.com/openego/dingo/commit/6f2158f520524c71102ae739da2aa18f2c2ab7af#L25-#L87) and [here](https://github.com/openego/dingo/blob/16f3455bb4af314e6c949e55ae7a0cf9c9f8b751/examples/example_multiple_grid_districts.py).

TODO
1. Identify additional locations in the code where (a) home dir is created or (b) needs to be created
2. Move this code a more central place, i.e. to the setup-process. Would be nice to create the directory when dingo is installed
3. Divide [connection()](https://github.com/openego/dingo/commit/6f2158f520524c71102ae739da2aa18f2c2ab7af#L25-#L87) in multiple modular functions: (a) create dir(s), (b) check for config, (c) create default config file.
4. Consider required subdirectories when creating dingo home dir

nesnoj, 2017-06-01:
Dublette #181

piaulous, 2023-03-13:
Closed, is doublet.

closed 2023-03-13 by piaulous