openego/ding0 #178: Statistical analysis of Dingo grid topology data

gplssm, 2017-05-29:
There are several motivations for performing a statistical evaluation of Dingo grid topology data.

1. Evaluate results of recents features addtition (see #169 )
2. Assess quality of input data: for example data on location of RES generators can be evaluated by analysing grid connection points
3. Provide potential users on  overview on available data

nesnoj, 2017-06-02:
Please consider #186, #122

gplssm, 2017-06-06:
Implementation of calculating key figures is separated into two steps: one for MV (#189) another for LV (#190).

gplssm, 2017-07-17:
**List of statistical evalutation aspects** _(to be extended)_

1. Are voltage levels at nodes exceeding limits?
2. Are branches overloaded?

gplssm, 2017-07-22:
Closed in favor of #212

closed 2017-07-22 by gplssm