openego/ding0 #181: general place for files

nesnoj, 2017-05-31:
We got different places for input- and output files.

Example: Currently in `save_nd_to_pickle()` the pkl is exported to script path (example dir) if no path is given. Should we adjust the defualt export path to `~/.dingo/output/` or similar?

If we want to solve the where-do-we-read-and-write-files-from/to-problem in a more general way we should make a list of all files we use. I looked up some stuff:

- [animations](https://github.com/openego/dingo/blob/397bb8991877b9b4e8e4a181bdad95b5b06dc11b/dingo/grid/mv_grid/solvers/base.py#L99): `dingo/output/animation/`
- report: `dingo/output/report/` (where is it used?)
- [debug](https://github.com/openego/dingo/blob/397bb8991877b9b4e8e4a181bdad95b5b06dc11b/dingo/tools/debug.py#L23): `dingo/output/debug/` (I'm to check if this one is still needed)
- [pickle](https://github.com/openego/dingo/blob/397bb8991877b9b4e8e4a181bdad95b5b06dc11b/dingo/tools/results.py#L132): `<CURRENT DIR>`
- [pyPSA](https://github.com/openego/dingo/blob/397bb8991877b9b4e8e4a181bdad95b5b06dc11b/dingo/tools/pypsa_io.py#L324): 
- ...

Could you add more @gplssm ?

This leads to more issues:

1. At start of DINGO, check if config dir and subdirs/files exists, if not create it (similar to [this way](https://github.com/openego/dingo/blob/397bb8991877b9b4e8e4a181bdad95b5b06dc11b/dingo/tools/db.py#L25)? https://github.com/openego/dingo/issues/140#issuecomment-304143179)
2. Store dir definition in config_files.cfg
3. ... (insert more ideas here)

Furthermore, we might discuss how to handle config files: The standard-config could be overridden by custom config placed in `~/dingo/...` (nice-to-have :nerd_face:).

nesnoj, 2017-06-01:
Dublette #177

mltja, 2023-03-13:
Consider also mv and lv plots

closed 2026-10-30 by nesnoj