openego/ding0 #182: Docstring Check of dingo/tools

miguelparadacontzen, 2017-05-31:
Please check docstrings of dingo/tools. Some descriptions are missing and some types of variables should be confirmed or added. There are several documents but it shouldn't take too much time to go through them. 

The parameters of a function are documented like:

`Parameters`
`----------`
`name: type`
`____Short Descr`

After the section title "Parameters" comes a line "--------" with the same number of - as letters in the title. Equivalent to Parameters are Arguments and Args
the parameters names and type come afterwards without indentation. The description of each variable comes in the following line WITH indentation.  type and description are optional but desired.

The returns have a different syntax:

`Returns`
`-------`
`type`        
`____Description of returns`

After the type (without indentation) comes the description of the return WITH indentation. The name of the variable that is returned is not documented. If the function has no return, there is no need for the Returns section. type is NOT optional.

gplssm, 2017-06-20:
I left some TODOs for you @miguelparadacontzen. Could you please check in addition if this part of the documentation still looks as expected?

Note, the file pypsa.io is still not done. Editing of results.py will be done directly in dev as I completely restructured it,

gplssm, 2017-06-21:
pypsa_io.py is done now. @miguelparadacontzen please go on...

miguelparadacontzen, 2017-07-13:
Updated. only the file results.py have some parts to be completed.

Also, there is a bib reference to a “DENA Verteilnetzstudie” that needs to be specified.

commit 218e21a8 (2017-07-18): Fix #182

I have edited the reference.

Fine, I close it :tada: