openego/ding0 #195: Regression test(s) for Dingo results data

gplssm, 2017-06-07:
Implement a so-called regression test that takes results previously generated and compares these with results from a new run.
Therefore, we take 2 Dataframes (nodes_df, edges_df) that are generated by `nd.to_dataframe()` and compare these via [pandas equals function](https://pandas.pydata.org/pandas-docs/stable/generated/pandas.DataFrame.equals.html). How the `to_dataframe` function is used, see [this example](https://github.com/openego/dingo/blob/957f26f14bd2488d2d6247478ba2f34560636932/dingo/examples/example_analyze_single_grid_district.py#L42).

A potential roadmap of implementation
1. Make sure [pandas equals function](https://pandas.pydata.org/pandas-docs/stable/generated/pandas.DataFrame.equals.html) works properly with `nodes_df` and `edges_df`
2. Create test directory and a test file (for inspoiration about structure see resources below)
3. Add exemplary results file to github repository
4. Write test function itself that compares the save results data with the newly generated data from a Dingo run
5. Check if (a) test runs successfully if data is the same and (b) test fails if data is different

Resources for tests in Python
* Summary on tests in python in the [Hitchhikers guide](https://python-guide-pt-br.readthedocs.io/en/latest/writing/tests/)
* Regression tests on the example of [factorial](https://blog.udemy.com/python-unittest/)
* How [oemof](https://github.com/oemof/oemof/blob/d7b85117825bc824d5794ac972982efede791b4b/tests/regression_tests.py) tests against regression

Note: Please implement this feature in a feature branch that is branched from latest dev state.

Hint: As far as I know, nosetest is the preferred python package for testing.

nesnoj, 2017-06-08:
Keep in mind that there's already a function to compare two graphs [here](https://github.com/openego/dingo/blob/b319e8ab06231e70da307593594a116dcbced701/dingo/tools/debug.py#L23).

miguelparadacontzen, 2017-06-13:
Question:

is dingo deterministic?

i.e. two different runs of dingo on the same input, always give the same output?

For example, in the code
` # database connection`
` conn = db.connection(section='oedb')`
` # instantiate new dingo network object`
` nd = NetworkDingo(name='network')`
` # choose MV Grid Districts to import`
` mv_grid_districts = [3545]`
` # run DINGO on selected MV Grid District`
` nd.run_dingo(conn=conn, mv_grid_districts_no=mv_grid_districts)`
` # New run from dingo`
` # instantiate new dingo network object`
` nd_dos = NetworkDingo(name='network')`
` # run DINGO on selected MV Grid District`
` nd_dos.run_dingo(conn=conn, mv_grid_districts_no=mv_grid_districts)`

should nd and nd_dos always be the same?

nesnoj, 2017-06-14:
Yes and no :).

The relevant content, that is the topology of the graph (nodes and edges) should be deterministic. But if you would compare two ND or a node which is the same with regard to contents, e.g. the HV-MV station `nd.mv_grid_districts[0].mv_grid.station` you'll get `False`.

Background info: You can compare if two objects referred to by two variables are equal by using `a == b` whereas `a is b` tests if both variables point to the same object.
Example:
```
>>> a = 5
>>> b = a
>>> a == b
True
>>> a is b
True
>>> b = 5
>>> a == b
True
>>> a is b
False
```

To test equality of a graph we neither can use `==` nor `is` due to complex structure of the objects (we could define these two operators but want to avoid this fancy stuff ;). Instead, as a simple solution to check for determinism of topology we can check if all nodes and edges in a `GridDingo._graph` are contained in a second one and the counts are equal. See method linked above. A string representation is given by `repr()` ([example def](https://github.com/openego/dingo/blob/ebd94501697f22ff5fc00dc8809e1d3bef899fcf/dingo/core/network/stations.py#L180-L181)).

gplssm, 2017-06-22:
I talked to @nesnoj about the IDs of branches.

It cannot be guaranteed that these are the same in different runs of Dingo. The same IDs should result of differnts runs, but they don't have neccessarily point to the same objects. Thus, a comparison of branches along the IDs is not possible.
@nesnoj suggestion is to compare branches by their neighboring nodes. A branch should have the same neighbors in different runs.

@miguelparadacontzen if you need further information, we can discuss in the office.

gplssm, 2017-07-17:
Thanks for the implementation of this features!

Before we pull this feature into dev, please do the following minor changes.
- [ ] Delete of commented/ not used code
- [ ] Move the file `test_dingoruns.py` to to `dingo/tools/tests.py`. 
- [ ] Provide an example of how to use this features in the [developer notes](https://github.com/openego/dingo/blob/670cb40982bb218497fbf460a1d703db16664a80/doc/development.rst). Use a new section.
- [ ] Add docstrings where required
- [ ] Do we need the pickle file in the repository? If we replace the frequently they blow up the repository size

miguelparadacontzen, 2017-07-17:
done

gplssm, 2017-07-19:
Thanks! Looks good now! I go for the merge...

closed 2017-07-19 by gplssm