openego/ding0 #200: Issues according to tests

miguelparadacontzen, 2017-06-27:
While running tests in the reg-tests branch, I have found the following issues:

1. Function `to_dataframe` does not sort its outcome. So two different runs over the same network, will typically give two differently ordered outputs. A workaround is already implemented in the branch, but the sorting criteria for the lines is its `'length'` attribute.

2. The IDs of the lines/branches are not unique in different runs of dingo. 

3. The adjacent nodes of a branch might be inverted in different runs of dingo. Therefore, the `'geom'`attribute of the dataframe is not necessarily the same one. 

4. Different runs of dingo set the values of the `'s_res'`attributes of the branches and the `'v_res'`attributes of the nodes to different values!! Rounding them to 5 decimal places seems to solve the problem for the nodes, but not for the branches. 

5. The most critical thing, is that, some times, two different runs of dingo give solutions with a different number of nodes! I have not testes what kind of nodes are they, but in the cases that this happens there is one extra node.

nesnoj, 2017-07-13:
Concerning 5.:
I tried to track down but unsuccessfully until now.
The issue seems to be related to connection of MV generators in [`mv_connnect()`](https://github.com/openego/dingo/blob/7848c10197b825f8ffba39681cbb59016323ea67/dingo/grid/mv_grid/mv_connect.py#L808). To go on your can checkout the branch `features/node_finder` which contains some debugging stuff.

nesnoj, 2017-08-21:
5. was solved in https://github.com/openego/ding0/commit/0ef1253f383482e32d76e86fa96a4bf3c40c045c, thank you @miguelparadacontzen!

What of the problems 1-4 have not been solved yet? @miguelparadacontzen

closed 2017-08-25 by miguelparadacontzen