openego/ding0 #201: MV grid outliers

nesnoj, 2017-06-27:
We have different grids with quite long total length from HV-MV subst. to farthest MV-LV subst. (ring(s) are open) with a max. of ~84km (MVGD 3093). We might add an additional constraint on max. half-ring length in `tech_constraints_satisfied()`.
To evaluate the need, a histogram of max. distances should be created first (see #189).

nesnoj, 2017-07-24:
This example was caused by the circ. breaker relocation using loads as well as generators.
From 1046681149ba9358056df97cd684572cc7ce8070 (see also #167) on the standard value is loads only, so this MVGD should be ok now.
@gplssm Please create a new grid dataset as soon as I give you a go (I wanna fix sth else first)

gplssm, 2017-07-24:
Yep, I'll do that. I did a new run on the weekend and it took only 24 hrs...

nesnoj, 2017-07-24:
:ok:, this one is done too. U can go for it

closed 2017-07-24 by nesnoj

nesnoj, 2017-07-24:
Update - what I did:
I added an additional constraint for the max. ring length of `2 * v_level * max_ring_length_factor` (e.g. in a 20kV grid assuming max_ring_length_factor=1.5 (default in `config_calc`) the max ring length would be 60km) . `max_ring_length_factor` is used to achieve more variance in the grids.

@jochenbuehler Is a factor of 1.5 reasonable?

jochenbuehler, 2017-07-26:
Seems so:
[image]

nesnoj, 2017-08-21:
See #224 for new implementation

commit 61f70218 (2017-09-06): add additional constraint on max. ring length, fixes #201