openego/ding0 #206: get info if grid is unstable

nesnoj, 2017-06-29:
There's no way to notice whether a created MV grid is stable or not except by reading the log (line is written [here](https://github.com/openego/dingo/blob/ee237e37d4c228081e1e246d7e6d0d431c6dda9e/dingo/flexopt/reinforce_grid.py#L81)). We need a conspicuous warning and a flag in MVGridDistrictDingo, a list of unstable grids in NetworkDingo or similar!
Additionally, we could add an additional parameter/option (e.g. `stable_only`) to the pickle export method to control if only stable grids are accepted..
@gplssm What do u think?

gplssm, 2017-06-29:
Sounds like a nice feature!

My plan was actually to do this via the statistical analysis: export the number of unstable nodes/lines in a grid or similar.

gplssm, 2017-07-22:
Closed in favor of #212

closed 2017-07-22 by gplssm