openego/ding0 #209: Extend export of results to OEDB

gplssm, 2017-07-17:
Currently [export_mv_grid_new()](https://github.com/openego/dingo/blob/670cb40982bb218497fbf460a1d703db16664a80/dingo/core/__init__.py#L1222-#L1337) write basic information about generated grids. Extend the information exported to OEDB by data obtained from [nd.to_dataframe()](https://github.com/openego/dingo/blob/670cb40982bb218497fbf460a1d703db16664a80/dingo/core/__init__.py#L1345-#L1436).

ingaloeser, 2017-10-09:
Some attributes had different names for LV and MV, e.g. grid. I put the MV name in brackets. Is it possible, that some attributes were omitted for the MV definition or did I misunderstand it?

mltja, 2023-03-13:
Reopen if needed after discussion in #384

closed 2023-03-13 by mltja