openego/ding0 #213: Peak Load differences

miguelparadacontzen, 2017-07-24:
there are peak load attributes that are not consistent between them. For example, if we add all the peak loads contained in every LV_district in every Load Area in a MV_district, the value is different from that of the attribute MV_district.peak_load.

The inconsistency is possible also by the generation values. although I have not yet check that.

For example
'''
grid_id                        3545.0
peak_load                18750.509313
generation_capacity      11828.855000
biomass                    514.000000
solar                      307.000000
wind                      1800.000000
MV generation capacity    2621.000000
v_nom                       20.000000
LV generation capacity    9207.855000
LV peak load             18750.509313
km_cable                    34.891086
km_line                    165.340363
cable_%                     17.425378
line_%                      82.574622
rings                        6.000000
Cable distributor count     52.000000
LV Station count           146.000000
Unknown count                1.000000
biomass count                1.000000
solar count                  1.000000
wind count                   3.000000
Acc_smax_Trafos            126.000000
LV LA agg                    1.000000
LV LA agg_peakload       11788.988400
LV LA agg_pop            18516.000000
LV LA groups                24.000000
LV LA nor                   34.000000
LV LA sat                   28.000000
LV districts               216.000000
N_trafos                     2.000000
Peak Load Agricultural    8979.137767
Peak Load Industrial      5340.896236
Peak Load Residential    12716.383522
Peak Load Retail          3503.080188
'''

nesnoj, 2017-07-24:
What strikes here is if you sum up the Peak loads and subtract the MVGD peak load you get the peak load of agg. LA:
(8979.137767+5340.896236+12716.383522+3503.080188) - 18750.509313 = **11788.9884**
So in the statistic above the peak load of agg. LA seems to be excluded.. which code snippet did you use for your numbers? (where does the MVGD peak load comes from?) :electric_plug:

miguelparadacontzen, 2017-07-26:
Good observation!

The value 'peak_load 18750.509313' comes from the function 'to_dataframe()', while the others come from simply adding the attribute of every LA area in the district in a for cycle. 

I think that the difference comes because when the graph of the MV grid is built, the nodes that are an instance of class 'LVLoadAreaCentreDingo' are not actualized with the correct value. See ['to_dataframe()' function](https://github.com/openego/ding0/blob/dev/ding0/core/__init__.py) from line 1388 on.

nesnoj, 2017-08-18:
The problem is: In line [1372](https://github.com/openego/ding0/blob/74f8f8ee7ba1f9d87757773145c164bd61ff5522/ding0/core/__init__.py#L1372) we check if the node was part of the PF calculation (hence got attr. `voltage_res`) but neither the LVstations in agg. LA nor the LAcentres were. I fixed this in d04cb1169e425adcda2246ecf7bf01ccf2e3744a. I kicked the `hasattr(node, 'voltage_res')` and include ALL STATIONS, LA CENTRES AND SWITCH DISCONNECTORS in the result dataframe (**note: I added additional type strings => these have to be incorporated in evaluations ([example](https://github.com/openego/ding0/blob/d04cb1169e425adcda2246ecf7bf01ccf2e3744a/ding0/tools/results.py#L103))**).
Load and generation (at least in MVGD 3545) are correct now.

Could you please verify the results again with your script @miguelparadacontzen ?
@gplssm: Any objections? (I'm surely not aware of all consequences these medifications do have..)

If everything is fine, please close this issue..

gplssm, 2017-08-18:
@miguelparadacontzen could you please use the regression test here to approve that nesnoj's changes do not affect the results?

miguelparadacontzen, 2017-08-28:
The changes in function `to_dataframe()` do not have an impact on the calculation of the MV stats in `tools.results.py,` because we decided not to use it :)

Because of this issue, to avoid possible similar situations, and to make it easier to calculate the LV stats, we decided it was better to obtain the data "from below", looping over every MV node to loop again further over the LV districts within. So, at the end it was more efficient not to use the function, because I needed to loop over the branches and nodes anyway to obtain other figures that are not in `to_dataframe()`, and by using it I would need to loop twice over everything.

@gplssm : I don't think it is necessary to use a regression test to check. Unless `to_dataframe()` is used for something else in dingo, or in the process of building the network.

gplssm, 2017-08-28:
Thanks for explaining this again! Actually I should know ;-)

Shall we remove `to_dataframe()` from the code basis?

nesnoj, 2017-08-29:
If I get this right, the bug initially described does not occur in the new stats version?
If the old `to_dataframe()` is not used anymore (I found 1 usage in `concat_nd_pickles()` though), I don't mind to delete it.
Does it make sense to move your new stats version to a separate function?

gplssm, 2017-09-01:
Closed in favor of #233

closed 2017-09-01 by gplssm