openego/ding0 #214: Statistical analysis of the entire Dingo dataset (on MV level)

gplssm, 2017-07-25:


jochenbuehler, 2017-07-26:
Please verify the following values (distance between stations, maximum length of outgoing lines) with the Dingo dateset: 
[image]

jochenbuehler, 2017-07-26:
Also please verify these values (source: Brunekreeft G, Luhmann T, Menz T, Muller S and Recknagel P 2015 Regulatory Pathways For Smart Grid Development in China 1st ed (Springer Vieweg)):

[image]

jochenbuehler, 2017-07-26:
Here is more :)
[image]

(soure: http://www.nlt.de/pics/medien/1_1191828963/Kabel_oder_Freileitung_korr.pdf)

mltja, 2023-03-13:
Mentioned in #212 for information

closed 2023-03-13 by mltja