openego/ding0 #218: Number of rings does not match number of circuit breakers

abokam, 2017-07-27:
In most of MV grids No. of rings sometimes is double of no. of CBs and sometimes is more by 1, 2 or even 3. 
ex. of not equal:
MVGD 4 : No_rings =  2  & No_CB =  1
MVGD 193 : No_rings =  4  & No_CB =  2
MVGD 9 : No_rings =  3  & No_CB =  2
MVGD 59 : No_rings =  5  & No_CB =  2

ex. of equal:
MVGD 8
MVGD 13
MVGD 417

jochenbuehler, 2017-07-27:
I dont really understand it. Normaly every outgoing line of a HV MV subastation has a circuit breaker connectet to it. Therefore a ring has to have two circuit breakers...?

abokam, 2017-07-27:
What I meant by CBs is not the outgoing CBs from the substation I mean the switch disconnector that be placed in the lowest current region of a ring. It remains open at normal operation and what I understand is that every ring has to has a single disconnector.

nesnoj, 2017-07-28:
Right, every ring is supposed to have exactly 1 CB - I'll dive into this, thanks for reporting!

nesnoj, 2017-08-18:
It's not a bug, it's a feature :).
The explaination is quite simple: Aggregated LA do have a ring but no Circuit Breaker (=Switch disconnector) since we don't need it for our calculations.
Consequently, count(rings) = count(CB) + count(LA_aggregated)

closed 2017-08-18 by nesnoj

miguelparadacontzen, 2017-08-23:
Hey, does this mean that between the MV station and the aggregated LA there are always two branches? or one branch only?

is one of those the case for every agg LA? or there might be two LAs in a "ring"?

nesnoj, 2017-08-23:
Types in dingo's (which are counted as rings):
1. Regular: connects >=1 regular LA with 1 full / 2 half-rings, has 1 switch disconnector / circuit breaker
Caution: This type might looks like 2. if it has only 1 LA
2. Aggregated: connects **exactly** 1 agg. LA with 1 branch, has no switch disconnector / circuit breaker