openego/ding0 #221: Voltage and current for nodes and edges

gplssm, 2017-08-17:
Write function similar to other stats functions that exports voltage and current levels at nodes/edges. For both grid levels LV and MV. Thus 4 tables resuls here.

miguelparadacontzen, 2017-08-25:
I'm working on the functions, but I don't know what to do if the attributes ’v_res’ or ’s_res’ are not present. Is then the node or branch "wrong"? or that shouldn't happen at all?

I've found that in the MV districts that I have tested, LV branches don't have 's_res' at all.

nesnoj, 2017-08-25:
We do not use power flow calculation in LV grids, this is why nodes or branches do not have these attributes. MV nodes have v_res and branches have s_res, each is an array containing 2 values (heavy-load flow case and feed-in case).
These values should be exported (maybe using 1 or 2 extra dataframes?).

miguelparadacontzen, 2017-08-25:
what should I write in the LV dataframes then? @gplssm?

nesnoj, 2017-08-25:
Sorry, my bad! These attributes might exist at LV level too; instead of PF, we do a simplified calculation here. @gplssm?

gplssm, 2017-08-28:
We talked in natura. These attributes don't exist due to choice of implementation. Decision: Just export nodes/edges where issues remain.

nesnoj, 2017-08-29:
please close if resolved.

closed 2017-09-01 by gplssm