openego/ding0 #222: Parallelization of run_dingo and stats in example scripts

gplssm, 2017-08-17:
Parallelize a run of multiple grid districts. Based on the [example of multiple grid districts](https://github.com/openego/ding0/blob/2acbb7adf280d25153fd2a9561a3d6393f30e183/dingo/examples/example_multiple_grid_districts.py) implement parallelization.

For the stats, adapt the [current analysis script](https://github.com/openego/ding0/blob/2acbb7adf280d25153fd2a9561a3d6393f30e183/dingo/examples/example_analyze_multiple_grid_districts.py) to generate all stats (MV. LV, voltage, current) by using parallelization. For inspiration use [this gist](https://gist.github.com/gplssm/94733442ecd3f6ece979db0e1d4cf745#file-dingo_statistical_key_figures-py).

nesnoj, 2017-08-18:
Ok, then I'll close the duplicate #39.
;)

nesnoj, 2017-08-18:
I guess you also mean the use of >1 CPU cores?!

gplssm, 2017-09-01:
Thanks to @miguelparadacontzen :tada:

closed 2017-09-01 by gplssm

nesnoj, 2017-09-01:
Benchmark? :)

gplssm, 2017-09-01:
> Benchmark? :)

Later. Did just some tests on small subsets...

miguelparadacontzen, 2017-09-04:
Just to brag, I run dingo over all MV districts (1 to 3607), on two cores, and saved all stats (MV stats, MV critical nodes, MV critical branches, LV stats, LV crit nodes, and LV crit branches) and it took 55964.686014175415 seconds = 15,54 hours.

nesnoj, 2017-09-14:
The parallelization did not work if count of 2*MVGDs <= count of cpu cores (e.g. [here](https://github.com/openego/ding0/blob/e245686ff57a13734cbc365ec8b780d95259b0c4/ding0/examples/example_parallel_multiple_grid_districts.py#L83)).
(I tried 4 MVGDs on 8 cores).
I fixed this in `results.py` and `example_parallel_multiple_grid_districts.py` in branches `master`, `dev` and `features/stats-export`. Please fix other occurences, if existing.

Hmm